Smartphone-Companions/ESP32-ANCS-Notifications #1: Handle reconnection when Bluetooth is disabled

jhud, 2019-12-20:
The library should start advertising again if the Bluetooth drops out.

Petros144, 2019-12-22:
Hi Jhud!

Thank you for the great implementation of ANCS! I played a bit with it and if i disable BT on my iPhone and enable it again, it wont reconnect automaticly.

is there a solution for this problem?

jhud, 2019-12-22:
> Hi Jhud!
> 
> Thank you for the great implementation of ANCS! I played a bit with it and if i disable BT on my iPhone and enable it again, it wont reconnect automaticly.
> 
> is there a solution for this problem?

Hi, Petros, thanks! That's what this bug is about... I'm going to look at any and all reconnection issues soon.

closed 2019-12-23 by jhud