Smartphone-Companions/ESP32-ANCS-Notifications #5: Add Android Support

Petros144, 2020-02-20:
Hi Jhud!

I would really like to implement the Android notifications into this Library, I did some research and found something:

https://github.com/mitchwongho/ESP-Alerts-for-Arduino

Galaxy-Man, 2020-03-06:
The answer is yes, it takes awhile to update the Android Studio. I have mine compiling with minimal warnings showing deprecated items which need to be fixed. Other than that it works fine.
To make use of the Android and ESP a programmer needs to learn both IDEs. 
There is no quick root, just practice practice practice.

Galaxy-Man, 2020-03-06:
One issue I am having is the messages are show ok but when a phone call comes into the device I get this

onWrite() {uuid=8d8218b6-97bc-4527-a8db-13094ac06b1d,len=29}
Phone:"null" via Call @ 15:28

Is there a setting on the Android phone I need to set to allow the call detail to show where null is?

My phone is a Galaxy S9 with Android 9, all notifications in settings are turned on, unless I am missing something??

Petros144, 2020-03-06:
hi! What i found is, that some Apps will not show any Notifications such Music app, than I get the Null error...

I tested the calls, and all went well. I see caller and incomming call. what app do you use to make calls? 

I mailed Jhud and he also wants to realease a Android version as well but this will take some time (months). so for now we stuck with the ESP Alert app.

Galaxy-Man, 2020-03-09:
Hi, currently using the native dialer on Samsung Android, I will try another pone and also hard reste this device to prove a point. All other messages work for full length and are not truncated. I think there must be a setting for the phone app such as security permissions etc but where it is I am not sure. 

What phone are you using?

I am also trying to change the MTU requests on Andoird as that is the client and therefore dont need ot set on ESP32.

I would appricate any help on the Android issue with the phone null response.

Petros144, 2020-03-09:
So i see, You compiled your own APK data right? im useing the original file provided in my first post and it works well on older and Newer Phones. I test mine with old M8 HTC (im iPhone user) and try all imprntant apps that I need. Maybe you can provide your APK so I can test also...?

Just give it a shot with my APK and test the Call.

jhud, 2020-03-11:
Hi! Thanks for the code, @Petros144 . The long-term plan would be to include Android transparently as part of the same library, based on the same sort of code you uploaded. And we would also have a simple example Android app to go with it that builds as easily as possible; needing an app is an additional complication for Android that we'd want to make as painless as possible.

This library is about to be used heavily by an institution over the next few months, and we'll take the next steps based on feedback from that - which may include Android support.

Galaxy-Man, 2020-03-11:
Hi Petros

Your APP does the same, NULL in the phone details. Do you know how to use the debug in Android for the device?

Petros144, 2020-03-11:
@jhud Thank you for the Email & reply. Hope this will be done in the next release! @Galaxy-Man what debug mode you mean? in ESP Code or Android device?

Galaxy-Man, 2020-03-12:
Debug in Android

Petros144, 2020-03-12:
no sorry