Smartphone-Companions/ESP32-ANCS-Notifications #6: Reconnect-Issue - cycling bluetooth does not reconnect to the ESP32 on some devices

jhud, 2020-02-21:
tested it with the Lolin D32 with the latest board definitions and code and with both iPhone SE on iOS 12.3.1 and with iPhone 8 with iOS 13something. If you have an older iPhone with you and would be able to test it with that and see if we are doing something wrong or if it is really a problem with these phones that would be great. The issue for us is the following:  If we walk away with the iPhone or turn off bluetooth on the phone and back on it will not be able to reconnect. Even if we try to reconnect by tapping the device on the phone it wont be able to do it. The only way to reconnect it is then resetting the ESP32 and it reconnects immediately.

jhud, 2020-02-24:
I cannot reproduce this on iPhone X, iPhone 5S, TTGO or ESP-WROOM-32 devices with the latest version of the library. This is the output I get.

_Connect with iPhone X, using ESP-WROOM-32:_

ESP32-ANCS-Notifications Example
------------------------------------------
StateConnected - connected to a phone or tablet
StateDisconnected - disconnected from a phone or tablet
StateConnected - connected to a phone or tablet
Got notification: Engelbert Testa
test
com.facebook.Messenger
social
1

_Connect with iPhone 5S on same session, walking in an out of range, turning BT on and off on the phone multiple times:_

StateDisconnected - disconnected from a phone or tablet
StateConnected - connected to a phone or tablet
StateDisconnected - disconnected from a phone or tablet
StateConnected - connected to a phone or tablet
StateDisconnected - disconnected from a phone or tablet
StateConnected - connected to a phone or tablet
StateDisconnected - disconnected from a phone or tablet
StateConnected - connected to a phone or tablet
StateDisconnected - disconnected from a phone or tablet
StateConnected - connected to a phone or tablet
StateDisconnected - disconnected from a phone or tablet
StateConnected - connected to a phone or tablet

_TTGO ESP32 connect with iPhone 5S, walking in an out of range, turning BT on and off on the phone multiple times:_

ESP32-ANCS-Notifications Example
------------------------------------------
StateConnected - connected to a phone or tablet
StateDisconnected - disconnected from a phone or tablet
StateConnected - connected to a phone or tablet
StateDisconnected - disconnected from a phone or tablet
StateConnected - connected to a phone or tablet
StateDisconnected - disconnected from a phone or tablet
StateConnected - connected to a phone or tablet

alainbellet, 2020-02-24:
I updated all board definition and lib and i just tested now with two configurations and it seems to work! As you described (reconnecting after losing the connection, walked away, turn iphone on/off).
Iphone 8, iOS 13.3
Lolin D32 (ESP32 - WROOM 32)
Lolin D32 Pro (ESP32 - WROVER B)

So, already a good thing!

closed 2020-02-25 by jhud