Smartphone-Companions/ESP32-ANCS-Notifications #7: Still dependencies of Neil Kolban "ESP32 BLE Arduino"?

alainbellet, 2020-03-26:
I noticed that if the library "ESP32 BLE Arduino" is not installed i'm not able to compile the example code "ble_connection".

> Arduino: 1.8.12 (Mac OS X), Board: "LOLIN D32 PRO, Disabled, Default, 80MHz, 921600, None"
> 
> ***/ESP32-ANCS-Notifications-master/src/esp32notifications.cpp: In member function 'void BLENotifications::startAdvertising()':
> ***/ESP32-ANCS-Notifications-master/src/esp32notifications.cpp:147:24: error: 'class BLEAdvertisementData' has no member named 'setServiceSolicitation'
>      oAdvertisementData.setServiceSolicitation(ANCSBLEClient::getAncsServiceUUID()); 
>                         ^
> exit status 1
> Error compiling for board LOLIN D32 PRO.

jhud, 2020-03-26:
ESP32 BLE Arduino is now part of the core Espressif libraries - you shouldn't need to install anything extra. You may have old versions of the libraries hanging around, so I updated the readme with that info.

alainbellet, 2020-03-27:
hey @jhud thanks for the update! Yes i know that ESP32 BLE Arduino is now part of the Core ESP32 Library, but i checked and i have the last version (1.0.4). And reinstalling the old ESP 32 Lib from Neil Kolban solve the compile error. (the problem came from a Student that had a fresh instal of the board definition as well as the ESP32-ANCS-Notification.
So i digged a bit more to point the problem and noticed that the class BLEAdvertisementData in BLEadvertising.h does not have the setServiceSolicitation in the packaged espressif Lib. But in Neil Kolban one it's present.. 

So i'm curious if you are able to compile without the Neil Kolban "original" one installed??

**Original**
> class BLEAdvertisementData {
> 	// Only a subset of the possible BLE architected advertisement fields are currently exposed.  Others will
> 	// be exposed on demand/request or as time permits.
> 	//
> public:
> 	void setAppearance(uint16_t appearance);
> 	void setCompleteServices(BLEUUID uuid);
> 	void setFlags(uint8_t);
> 	void setManufacturerData(std::string data);
> 	void setName(std::string name);
> 	void setPartialServices(BLEUUID uuid);
> 	void setServiceData(BLEUUID uuid, std::string data);
> 	void setShortName(std::string name);
> 	**void setServiceSolicitation(BLEUUID uuid);**
> 	void        addData(std::string data);  // Add data to the payload.
> 	std::string getPayload();               // Retrieve the current advert payload.
> 
**Included one**

> class BLEAdvertisementData {
> 	// Only a subset of the possible BLE architected advertisement fields are currently exposed.  Others will
> 	// be exposed on demand/request or as time permits.
> 	//
> public:
> 	void setAppearance(uint16_t appearance);
> 	void setCompleteServices(BLEUUID uuid);
> 	void setFlags(uint8_t);
> 	void setManufacturerData(std::string data);
> 	void setName(std::string name);
> 	void setPartialServices(BLEUUID uuid);
> 	void setServiceData(BLEUUID uuid, std::string data);
> 	void setShortName(std::string name);
> 	void        addData(std::string data);  // Add data to the payload.
> 	std::string getPayload();               // Retrieve the current advert payload.
>

jhud, 2020-03-27:
Thanks for the clarification. I thought I overloaded that class to avoid any dependence on that method being there or not, but I’ll have a look and update the docs/source if needed.

jhud, 2020-03-28:
Turns out my core library was patched with the extra setServiceSolicitation() method. This is fixed - this library now use its own implementation of setServiceSolicitation(). The API remains unchanged, but there is a #define switch in the cpp file if anyone wants to revert to nkolban's library. Or if the method moves to the Espressif core library, we can switch back to the core bluetooth implementation.

closed 2020-03-28 by jhud