Smartphone-Companions/ESP32-ANCS-Notifications #8: Add to library manager?

jhud, 2020-04-17:
We should host this on the official package list.

jhud, 2020-06-05:
Instructions are here: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

m-hertig, 2020-06-05:
Thanks jhud! I just followed the instructions and submitted the library here: https://github.com/arduino/Arduino/issues/10314

jhud, 2020-06-05:
Ah, I just saw - it was on my to-do list, but great!

m-hertig, 2020-06-05:
sorry, we were not sure on whos todo-list it was so I went for it

jhud, 2020-06-05:
No worries, as long as you are ready for it to go live - I was just waiting to see if anyone found any issues with it. Glad it should be on the official list soon!

alainbellet, 2020-07-31:
I just tested it now, it's available via the library manager, great! Now we should check that we keep it updated and push the updates to the manager repo. @m-hertig what was the procedure for that? We should also number well the versions as it allows for upgrade and downgrades via the manager.

[image]

closed 2020-11-19 by alainbellet