Smartphone-Companions/ESP32-ANCS-Notifications #11: Reconnect Issue

CoretechR, 2020-06-15:
Thank you for making this library!
Unfortunately I can't get the ESP32 to reconnect after being disconnected from an iOS device.
This has actually been a long discussed issue here: https://github.com/nkolban/esp32-snippets/issues/430

CoretechR, 2020-06-24:
It seems like the reconnect is working now on a different ESP32. Maybe it was a hardware related problem.

CoretechR, 2020-06-27:
When I start advertising after running notifications.stop(); the ESP32 cannot establish a connection anymore.
In order to get the WiFi to work I need to stop the BLE connection.
This is a simple example:
```
#include "esp32notifications.h"
#include <WiFi.h>
#include <HTTPClient.h>

// Create an interface to the BLE notification library
BLENotifications notifications;

// Holds the incoming call's ID number, or zero if no notification
uint32_t incomingCallNotificationUUID;

// This callback will be called when a Bluetooth LE connection is made or broken.
// You can update the ESP 32's UI or take other action here.
void onBLEStateChanged(BLENotifications::State state) {
  switch(state) {
      case BLENotifications::StateConnected:
          Serial.println("StateConnected - connected to a phone or tablet"); 
          break;

      case BLENotifications::StateDisconnected:
          Serial.println("StateDisconnected - disconnected from a phone or tablet"); 
          notifications.startAdvertising();
          break; 
  }
}

// A notification arrived from the mobile device, ie a social media notification or incoming call.
void onNotificationArrived(const ArduinoNotification * notification, const Notification * rawNotificationData) {
    Serial.print("Got notification: ");   
    Serial.println(notification->title); // The title, ie name of who sent the message
    Serial.println(notification->message); // The detail, ie "be home for dinner at 7".
    Serial.println(notification->type);  // Which app sent it
    Serial.println(notifications.getNotificationCategoryDescription(notification->category));  // ie "social media"
    Serial.println(notification->categoryCount); // How may other notifications are there from this app (ie badge number)
    if (notification->category == CategoryIDIncomingCall) {
		// If this is an incoming call, store it so that we can later send a user action.
        incomingCallNotificationUUID = notification->uuid;
        Serial.println("--- INCOMING CALL: PRESS A TO ACCEPT, C TO REJECT ---"); 
    }
    else {
        incomingCallNotificationUUID = 0; // Make invalid - no incoming call
    }
}

// Standard Arduino function which is called once when the device first starts up
void setup() {

    Serial.begin(115200);

    Serial.println("ESP32-ANCS-Notifications Example");
    Serial.println("------------------------------------------");    

    // Set up the BLENotification library
    notifications.begin("Memo");
    notifications.setConnectionStateChangedCallback(onBLEStateChanged);
    notifications.setNotificationCallback(onNotificationArrived);

    delay(3000);

    notifications.stop();
    delay(100);
    notifications.startAdvertising();
    
}

// Standard Arduino function that is called in an endless loop after setup
void loop() {   

}
```
I am gettings this output:
```
ESP32-ANCS-Notifications Example
------------------------------------------
[E][BLEAdvertising.cpp:259] stop(): esp_ble_gap_stop_advertising: rc=259 Unknown ESP_ERR error
[E][BLEAdvertising.cpp:157] setAdvertisementData(): esp_ble_gap_config_adv_data_raw: 259 Unknown ESP_ERR error
[E][BLEAdvertising.cpp:228] start(): << esp_ble_gap_config_adv_data (Scan response): rc=259 Unknown ESP_ERR error

```
Am I doing something wrong?

dfleck, 2021-02-10:
I am also having a heck of a time getting a reconnect after the device leaves range and comes back in. I'm running the sample code with no modifications. It will not automatically reconnect - and will not reconnect if I try to in Bluetooth settings. Only resetting will cause a reconnect. Here is what happens when it disconnects:

> [D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... Unknown
[I][esp32notifications.cpp:66] onDisconnect(): Device disconnected
[I][esp32notifications.cpp:144] startAdvertising(): startAdvertising()
[D][BLEDevice.cpp:556] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:196] start(): - no services advertised
[D][BLEDevice.cpp:556] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:196] start(): - no services advertised
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
[I][BLEDevice.cpp:604] removePeerDevice(): remove: 1, GATT role client
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
[D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEDevice.cpp:556] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 17]
[I][BLEAdvertising.cpp:507] handleGAPEvent(): STOP advertising
[D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEDevice.cpp:556] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 4]
[D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEDevice.cpp:556] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 1]
[D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEDevice.cpp:556] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 1]
[D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
[D][BLEDevice.cpp:556] getAdvertising(): get advertising
[D][BLEAdvertising.cpp:491] handleGAPEvent(): handleGAPEvent [event no: 6]
[D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown

This in on an iPhone SE2 with iOS 14.4.

Petros144, 2021-04-15:
@dfleck I get the same problem, it gets to the pint where the ESP restarts until you cut the power....  any ideas?

dfleck, 2021-04-15:
> @dfleck I get the same problem, it gets to the pint where the ESP restarts until you cut the power.... any ideas?

I see someone else has had reconnect problems and they have the same markings on their chip as I do. Using a different chip fixed their problem, so I have some new modules on order.

I have had restart problems, but they were due to bad power (wiggling the USB connector made them go away).

Petros144, 2021-04-19:
@dfleck I got a couple of modules with different Chips, I also made my PCB by myself so Power distribution should be no problem.
Still got the issue.

dfleck, 2021-04-23:
Even with new modules I still have the problem. If I walk my phone out of range and come back it never reconnects until I power cycle the ESP. Even selecting the device in the iPhone Bluetooth settings doesn't make it reconnect. This is with the sample code.

Does SerialMonitor give any indication what's going on (when I had power problems it kept giving me brownout detect errors)?

Petros144, 2021-05-12:
Any Updates? I would like to tackle the Problem, but the BLE stuff is pretty new to me, so my coding skills are not enouth tho...

this is the Backtrace of my error:

Decoding stack results
0x40093788: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x400939b9: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400e915b: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 174
0x400e74bf: log_printf at C:\Users\Petros T\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-uart.c line 495
0x400e5455: BLEUtils::dumpGattClientEvent(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at C:\Users\Petros T\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEUtils.cpp line 1284
0x400e1c5d: BLEDevice::gattClientEventHandler(esp_gattc_cb_event_t, unsigned char, esp_ble_gattc_cb_param_t*) at C:\Users\Petros T\Documents\Arduino\libraries\ESP32_BLE_Arduino\src\BLEDevice.cpp line 153
0x4015dfd1: btc_gattc_cb_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/profile/std/gatt/btc_gattc.c line 31
0x40155f0e: btc_task at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/btc/core/btc_task.c line 110
0x4008fed1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

MaxPuig, 2021-05-15:
An ugly solution is to "software-restart" the ESP32 when disconnected from the iOS device. 
`ESP.restart()` (No libraries are necessary)

Note: A PIN/password to connect to the ESP32's Bluetooth might be nice for privacy, but I don't know how to do that right now.
```c
void onBLEStateChanged(BLENotifications::State state) {
  switch(state) {
      case BLENotifications::StateConnected:
          Serial.println("StateConnected - connected to a phone or tablet");
          break;
      case BLENotifications::StateDisconnected:
          Serial.println("StateDisconnected - disconnected from a phone or tablet"); 
          ESP.restart(); // THE NEW LINE
          break; 
  }
}
```

Petros144, 2021-05-15:
I will try this, but i would really like to have a permanent working solution @MaxPuig

Sladerix, 2022-01-25:
Hi, is there a solution to the recconneting problem?

Petros144, 2022-02-09:
> Hi, is there a solution to the re-conneting problem?

I dont thnink so, only to restart the ESP...

ThreepE0, 2022-12-09:
I ended up putting ESP.restart(); in the StateDisconnected: check.  This way it restarts and advertising continues until you come back into range.  Not clean, probably not the best way, but it works for me