Smartphone-Companions/ESP32-ANCS-Notifications #14: No such file or directory found issue

ajoo3, 2020-08-04:
Hello, when I try compiling the code given here, I am met with this error:

In file included from /Users/Ajoo7530/Documents/Arduino/libraries/ESP32-ANCS-Notifications-master/src/esp32notifications.h:6:0,
                 from /Users/Ajoo7530/Documents/Arduino/Smart Lens/ESP32-ANCS-Notifications-master/examples/ble_connection/ble_connection.ino:8:
/Users/Ajoo7530/Documents/Arduino/libraries/ESP32-ANCS-Notifications-master/src/ble_notification.h:9:10: fatal error: string: No such file or directory
 #include <string>
          ^~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Nano.

I checked the file itself and it shows "#include <string> and I don't know if that's caused by a missing library or something else.

ajoo3, 2020-08-04:
I guess my main question would be, does this not support arduino nano?

alainbellet, 2020-08-04:
> I guess my main question would be, does this not support arduino nano?

Hello @ajoo3 , no the library is using BLE to connect to devices and is designed to work specifically with ESP32 Chip. Arduino Nano has no BLE onboard and is based on an other CHIP.

ajoo3, 2020-08-04:
Would it work if I connected an HM-10 BLE to the Arduino Nano?

alainbellet, 2020-08-05:
> Would it work if I connected an HM-10 BLE to the Arduino Nano?

No, the library is made for the esp32 architecture and won't compile on an Arduino Nano. If you need absolutely to use a nano you could have an ESP32 Board connected to it and make them communicate via serial for example. But that might be a bit overkill as the ESP32 itself is already highly capable.