Smartphone-Companions/ESP32-ANCS-Notifications #15: Date/Time and Reconnect issues

walle515, 2020-09-16:
Hi, im trying to use this to make something similar to a smart watch, for lack of a better explanation. I tried adding the notification attribute to the example code, and after converting it to a string, i realized the time, date, and year were all different. every time i boot the ESP32, it has a different date. If you/anyone knows any fixes or suggestions, it would be much appreciated. Also, a way to get the time from connection. I dont understand the ANCS or BLE stuff as much, and im very thankful this code is available to the public.

Also, as a side note, when I first connect, the phone asks if i want to allow the notifications to go through. When i forget the device and reconnect, i dont get that question, or the phone doesnt pair. For my project, that wont be an issue, but just so you are aware.

Again, any help would be appreciated. 

Hardware is the ESP32. For now, i am just getting it to work with the computer serial.

walle515, 2020-09-27:
Hi again. I dont mean to be pushy, but ive been experimenting and found a couple things out.

First, I am still experiencing the reconnect issue from time to time. I have also found that if you leave the device connected and then upload code, when its finished uploading, it gets every notification that is unread all at once and begins displaying them. My emails are not always marked read, and they all stream past during this. I have had to forget the device and try to reconnect to get this to stop. Not sure if this is an iOS thing or just the program.

Second, I have been researching getting a constant or more accurate time from the iPhone without needing the notification time. I found that iOS phones have a "Current Time Service" built in. While i have tried to implement this code myself, it uses a BLEClient function and I cannot find a way to access the "pClient" that I know already exists in your code. Do you have any suggestions for me as far as accessing that client? 

Here is a link to the code I'm referring to, and the function in question is "updateRTC"
https://github.com/takeru/Arduino_ESP32_ANCS/blob/M5StickC_Watch_Contest/Arduino_ESP32_ANCS.ino

Thanks again for the open source ANCS library for the ESP32!

walle515, 2020-10-03:
Hey guys, for those of you who wanted to get the time from the iphone, i finally figured it out. Using the function created by Takeru in the link above, I was able to get the current time from the device. I modified it such that it would utilize the TimeLib.h library so the Arduino sketch could access the current time. The file "ancs_ble_client.cpp" was edited to include the function and call it upon creation of the ANCS client. The edited file is included in this post.

The sketch was the example from this library, where the device will constantly check the time and only print the time to the serial if the minute changes. I will also include the sketch.

To the creator of this repository, I would recommend making this change so others using this library can access the time of the phone. This would give the ability to make a smart watch or smart clock, but also would allow people to use the time to get a correct time for notifications being recieved

[ANCS_BLE_Time.zip](https://github.com/Smartphone-Companions/ESP32-ANCS-Notifications/files/5321172/ANCS_BLE_Time.zip)

PixelDud, 2021-08-10:
This is pretty cool, sadly I couldn't get it to work in conjunction with my RTC.

jeffxiaofu, 2022-09-05:
> Hey guys, for those of you who wanted to get the time from the iphone, i finally figured it out. Using the function created by Takeru in the link above, I was able to get the current time from the [...]

Thank you!!!