Smartphone-Companions/ESP32-ANCS-Notifications #23: Ignore buffered notifications

Sladerix, 2022-01-25:
First of all, great library.
I was wondering if there is a way to ignore the buffered notifications at the moment of connection, but consider only the ones that arrives from the connection onwards.

I thought about waiting a certain amount of time after connection and ignore the things that arrives, but it doesnt seems a reliable way to do it, because the "time" depends on the amount of buffered notifications.

Riegelbauer, 2022-02-04:
Hello, that would be interesting for me too. I'm excited about the library. However, my ESP32 restarts endlessly when I have too many old notifications on my iPhone.

Dennis650, 2022-02-04:
Hello, that would be interesting for me too. I'm excited about the library. However, my ESP32 restarts endlessly when I have too many old notifications on my iPhone.

vladkorotnev, 2022-05-11:
Something like this inside your new notification callback should filter existing ones out:

```cpp
if((notification->eventFlags & ANCS::EventFlags::EventFlagPreExisting) != 0) {
            ESP_LOGI(LOG_TAG, "Received PRE-EXISTING notification 0x%x, discard", notification->uuid);
            return;
        } 
```

However, due to how the library is put together, it will still request the message/title/etc. even for pre-existing notifications. Applying the similar condition somewhere in the receive chain to your local copy of the library could help if you need that though :-)