Smartphone-Companions/ESP32-ANCS-Notifications #25: Cannot connect to iPhone

vladkorotnev, 2022-05-02:
I'm running the sample code with PlatformIO (Arduino framework on nodemcu-32s with espressif32, on a ESP-WROOM-32 devkit), using an iPhone 7 with iOS 13.0.

Trying to debug I have modified the `ANCSBLEClient::setup` method as follows:

```cpp
void ANCSBLEClient::setup(const BLEAddress * address) {
	ESP_LOGI(LOG_TAG, "!! Setup Begin !!");
    BLEClient*  pClient  = BLEDevice::createClient();
    BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);
    BLEDevice::setSecurityCallbacks(new NotificationSecurityCallbacks()); // @todo memory leak?

    BLESecurity *pSecurity = new BLESecurity();
    pSecurity->setAuthenticationMode(ESP_LE_AUTH_REQ_SC_BOND);
    pSecurity->setCapability(ESP_IO_CAP_IO);
    pSecurity->setRespEncryptionKey(ESP_BLE_ENC_KEY_MASK | ESP_BLE_ID_KEY_MASK);
    ESP_LOGI(LOG_TAG, "!! Setup Connect !!");
    // Connect to the remove BLE Server.
    pClient->connect(*address);

	ESP_LOGI(LOG_TAG, "!! Setup Service !!");
```

However, looking at the logs, the Connect part of the code never returns:
[log.log](https://github.com/Smartphone-Companions/ESP32-ANCS-Notifications/files/8602438/log.log)

Eventually, usually when leaving Bluetooth settings, the firmware will then crash due to #21.

Does anybody have experience with such a problem?
Thanks

vladkorotnev, 2022-05-03:
Found a temp fix that works: 

If using PlatformIO, change your platform line  to `platform = espressif32@3.5.0`.
(anything newer than 3.5.0 will fail)

If using Arduino IDE, downgrade the ESP32 board to 1.0.6.

Exact reason is unclear but it works for me.

ThreepE0, 2022-12-02:
> Found a temp fix that works:
> 
> If using PlatformIO, change your platform line to `platform = espressif32@3.5.0`. (anything newer than 3.5.0 will fail)
> 
> If using Arduino IDE, downgrade the ESP32 board to 1.0.6.
> 
> Exact reason is unclear but it works for me.

I want to hug you!  Thank you so much for sharing, I just spent hours troubleshooting various issues, roadblock after roadblock.  This last one was the board version though, and now my notifications are working.  Thanks!