Smartphone-Companions/ESP32-ANCS-Notifications #27: ESP32 Newest Release (2.0 +) Breaks this library

HoopmanJ20, 2022-08-03:
Hello, I have  used this library for a litany of projects at this point and I really appreciate all of the work you put into it. I just tried updating the ESP32 via Board manager from 1.06 to 2.0, 2.1, etc. and it seems this library breaks with this move. The first issue is that FreeRTOS gets messed up and as a result I get an error like "  xTaskGetHandle is not available ". Any ideas on if this can be fixed? I would use 1.06, but I am trying to use a esp32 C3 and that board is only supported after 2.0. I appreciate any thoughts or fixes that I can implement and again thank you for making this library!

molotovec, 2022-10-17:
Hey! I've edited .. PlatformIO\Projects\NotificationESP32S3\.pio\libdeps\esp32-s3-devkitc-1\ESP32 BLE ANCS Notifications\src\ancs_ble_client.h
by replacing
//#include "FreeRTOS.h" // For asynchronous tasks
#include "freertos/FreeRTOS.h"
#include "freertos/task.h" 

also edited line 40
public:
	TaskHandle_t clientTaskHandle; by replacing xTaskGetHandle  to TaskHandle_t 

source of inspiration is here https://community.platformio.org/t/esp32-taskhandle/4321

ShiveStar, 2022-10-19:
OK Epic I will try this out! I appreciate you posting this ^

ShiveStar, 2022-10-19:
Unfortunately, I am still not getting success after those edits. I get the same error on both Arduino and Platform IO:error: 
error: invalid new-expression of abstract class type 'NotificationSecurityCallbacks'.

Any chance you could post your code on your GitHub or maybe email it to me?

dfleck, 2022-12-09:
My ancs_ble_client.h line 6 is changed to:
#include "RTOS.h" // For asynchronous tasks
With this one change I am able to build with ESP32 by Espressif V2.0.5 and FreeRTOS 10.5.0-0.
HOWEVER - it doesn't work. It doesn't show up under Bluetooth on my iPhone. In issue #25 the suggested fix to this is to downgrade to ESP32 version 1.0.6.

Eonassis, 2023-04-15:
after a lot of trying I manage to compile for ESP32 C3, but it presents an error when sending to the board, but at least it compiles without errors

and what helped me?

oficial documentation Apple ANCS

https://developer.apple.com/library/archive/documentation/CoreBluetooth/Reference/AppleNotificationCenterServiceSpecification/Appendix/Appendix.html#//apple_ref/doc/uid/TP40013460-CH3-SW1 https://developer.apple.com/library/archive/documentation/CoreBluetooth/Reference/AppleNotificationCenterServiceSpecification/Specification/Specification.html#//apple_ref/doc/uid/TP40013460-CH1-SW7

@ParcoMolo copy #erro class MySecurity : public BLESecurityCallbacks in:

https://github.com/nkolban/esp32-snippets/issues/793#issuecomment-1189008694

intructions for @chegewara replace path > /libraries/BLE/src in:

https://github.com/espressif/esp-idf/issues/3230#issuecomment-479256634

instructions for @squonk11 #erro ringbuf_type_t issue by replacing it with RingbufferType_t. in:

https://github.com/nkolban/esp32-snippets/issues/933#issuecomment-569625223

instructions for @vijitSingh97 #erro esp_ble_gap_update_whitelist in:

https://github.com/nkolban/esp32-snippets/issues/472#issuecomment-501418988

I uploaded the modifications made to my fork

https://github.com/Eonassis/esp32_ANCS

as i did:

1 - Download the files from this repository

2 - Delete the src folder from the path, in my case it was: C:\Users\userpc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\libraries\BLE

3 - Copy the src folder from the repository to: C:\Users\userpc\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.7\libraries\BLE

3 - Delete the ESP32_BLE_Arduino folder from the path, in my case it was: C:\Users\userpc\Documents\Arduino\libraries

My arduino IDE is: 2.0.4

My esp32 libary is: 2.0.7

The ESP32 C3 is still not working, but I can already compile and send the board, when I send it here there is an error in the processor

Eonassis, 2023-04-15:
my now erro is:

20:44:04.184 -> ESP-ROM:esp32c3-api1-20210207
20:44:04.184 -> Build:Feb  7 2021
20:44:04.184 -> rst:0x3 (RTC_SW_SYS_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
20:44:04.184 -> Saved PC:0x4038195c
20:44:04.184 -> SPIWP:0xee
20:44:04.184 -> mode:DIO, clock div:1
20:44:04.184 -> load:0x3fcd5810,len:0x438
20:44:04.184 -> load:0x403cc710,len:0x91c
20:44:04.184 -> load:0x403ce710,len:0x25b0
20:44:04.184 -> entry 0x403cc710
20:44:04.987 -> 
20:44:04.987 -> assert failed: xQueueGenericSend queue.c:832 (pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTaskGetCurrentTaskHandle())
20:44:04.987 -> Core  0 register dump:
20:44:04.987 -> MEPC    : 0x40381d9c  RA      : 0x4038959a  SP      : 0x3fcb11e0  GP      : 0x3fc90800  
20:44:04.987 -> TP      : 0x3fc85a9c  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
20:44:04.987 -> S0/FP   : 0x3fcb13d7  S1      : 0x00000000  A0      : 0x3fcb1244  A1      : 0x3fc92349  
20:44:04.987 -> A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc96000  
20:44:04.987 -> A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x3fcb1234  S3      : 0x00000001  
20:44:04.987 -> S4      : 0x3fcb1238  S5      : 0x40389c66  S6      : 0x00000000  S7      : 0x00000000  
20:44:04.987 -> S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
20:44:04.987 -> T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
20:44:04.987 -> MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
20:44:04.987 -> MHARTID : 0x00000000  
20:44:04.987 -> 
20:44:04.987 -> Stack memory:
20:44:04.987 -> 3fcb11e0: 0x3fcb626c 0x00000000 0x3c0e6930 0x4038eb14 0x000000ff 0x635f7462 0x6b5f6766 0x3fc92348
20:44:04.987 -> 3fcb1200: 0x00000000 0xffffffff 0xffffffff 0x00323338 0x3fcb6504 0x3fc92d20 0x3c0e6930 0x3fc9309c
20:44:04.987 -> 3fcb1220: 0x3c0e6a09 0x3fc92d30 0x3fcb120c 0x3fc92d34 0x3c0e6d00 0x3fc92348 0x00000000 0x00000000
20:44:04.987 -> 3fcb1240: 0x00000000 0x65737361 0x66207472 0x656c6961 0x78203a64 0x75657551 0x6e654765 0x63697265
20:44:04.987 -> 3fcb1260: 0x646e6553 0x65757120 0x632e6575 0x3233383a 0x78702820 0x75657551 0x703e2d65 0x61654863
20:44:04.987 -> 3fcb1280: 0x3d212064 0x76282820 0x2064696f 0x2930292a 0x207c7c20 0x75517870 0x2d657565 0x782e753e
20:44:04.987 -> 3fcb12a0: 0x616d6553 0x726f6870 0x4d782e65 0x78657475 0x646c6f48 0x3d207265 0x2828203d 0x64696f76
20:44:04.987 -> 3fcb12c0: 0x30292a20 0x7c7c2029 0x51787020 0x65756575 0x2e753e2d 0x6d655378 0x6f687061 0x782e6572
20:44:04.987 -> 3fcb12e0: 0x6574754d 0x6c6f4878 0x20726564 0x78203d3d 0x6b736154 0x43746547 0x65727275 0x6154746e
20:44:04.987 -> 3fcb1300: 0x61486b73 0x656c646e 0x00292928 0x313d2530 0x00000000 0x00000003 0x3fcb61bc 0x3fca6eec
20:44:04.987 -> 3fcb1320: 0x00000000 0x00000000 0x3fcb16f8 0x40389c66 0x00000000 0x00000000 0x00000000 0x00000000
20:44:04.987 -> 3fcb1340: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000003 0x3fcb61bc 0x3fc96000
20:44:04.987 -> 3fcb1360: 0x00000000 0x3fcb60fc 0x3fc96000 0x42007048 0x3fcb35b8 0x00000000 0x00000001 0x00000001
20:44:04.987 -> 3fcb1380: 0x4038bb68 0x4038bb52 0x3fcb1400 0x313d2530 0x3fc85a9c 0x3fcb141c 0x3fcb60a8 0x42005e30
20:44:04.987 -> 3fcb13a0: 0x3fcb0590 0x00000000 0x00000001 0x00000001 0x00000000 0x00000001 0x600c0000 0x3fc96000
20:44:04.987 -> 3fcb13c0: 0xffffff9f 0x00000001 0xffffffff 0x3fcb05b4 0x00000000 0x00000000 0x00000000 0x313d2530
20:44:04.987 -> 3fcb13e0: 0x00000000 0x00000000 0x00000000 0x00000001 0x00000003 0x3fcb141c 0x00000000 0x42003340
20:44:04.987 -> 3fcb1400: 0x3fcb0120 0x3fc96000 0x3fcb6330 0x4201e49e 0x00000000 0x00000000 0x3fcb0168 0x00000000
20:44:04.987 -> 3fcb1420: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fcb0120 0x3fcb62c4 0x3fc9ab80 0x313d2530
20:44:04.987 -> 3fcb1440: 0x3fcb0120 0x00000000 0x3fcb6330 0x42047128 0x3fcb0120 0x00000000 0x3fcb0110 0x4204b1c8
20:44:04.987 -> 3fcb1460: 0x00000000 0x00000000 0x420470ee 0x3fcb6330 0x00000000 0x00000000 0x00000000 0x00000000
20:44:04.987 -> 3fcb1480: 0x00000000 0x00000000 0x00000000 0x4038ba96 0x00000000 0x00000000 0x00000000 0x00000000
20:44:04.987 -> 3fcb14a0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678
20:44:04.987 -> 3fcb14c0: 0x00000160 0xabba1234 0x00000154 0x3fcb1060 0xd3f9dbd8 0x3fc945b4 0x3fc945b4 0x3fcb14cc
20:44:04.987 -> 3fcb14e0: 0x3fc945ac 0x00000006 0x3fcb05bc 0x3fcb05bc 0x3fcb14cc 0x00000000 0x00000013 0x3fcb06bc
20:44:04.987 -> 3fcb1500: 0x5f435442 0x4b534154 0xe4ffc600 0x009328a3 0x00000000 0x3fcb14b0 0x00000013 0x00000000
20:44:04.987 -> 3fcb1520: 0x00000000 0x00000000 0x00000000 0x3fc9b584 0x3fc9b5ec 0x3fc9b654 0x00000000 0x00000000
20:44:04.987 -> 3fcb1540: 0x00000001 0x00000000 0x00000000 0x00000000 0x420bb1f2 0x00000000 0x00000000 0x00000000
20:44:04.987 -> 3fcb1560: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
20:44:04.987 -> 3fcb1580: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
20:44:04.987 -> 3fcb15a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
20:44:05.113 -> 3fcb15c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
20:44:05.113 -> 
20:44:05.113 -> 
20:44:05.113 -> 
20:44:05.113 -> ELF file SHA256: 6a70baba542a96f7
20:44:05.113 -> 
20:44:05.113 -> Rebooting...

Eonassis, 2023-04-16:
I get it not through the arduino IDE, I get it through the ESP-IDF security that everyone switches to it

https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/ble_ancs/README.md

biospank, 2023-05-30:
Hi,
I managed to fix compilation error for arduino_esp32 2.0.x library and successfully uploaded on a esp32-c3 board.
You can find a working version on my fork [here](https://github.com/biospank/ESP32-ANCS-Notifications)
The esp32-c3 board pairs with my phone but it doesn't show up dialog to allow permission for notifications.
I also noticed on serial monitor that on pairing I get this error log:

`BT_APPL: earlier enc was not done for same device`

Is anyone aware of the code to write/fix to resolve this issue?

Eonassis, 2023-06-18:
estou no mesmo local

> Olá, consegui corrigir o erro de compilação da biblioteca arduino_esp32 2.0.xe carreguei com sucesso em uma placa esp32-c3. Você pode encontrar uma versão de trabalho no meu fork [aqui](https://github.com/biospank/ESP32-ANCS-Notifications) A placa esp32-c3 emparelha com meu telefone, mas não mostra a caixa de diálogo para permitir permissão para notificações. Também notei nenhum monitor serial que, ao emparelhar, recebo este log de erro:
> 
> `BT_APPL: earlier enc was not done for same device`
> 
> Alguém está ciente do código para escrever/corrigir para resolver esse problema?

Salman-Mirza-0, 2023-06-26:
@biospank Have you managed to make further progress? I was able to use your fork on ESP32-S3. Right now, the board disconnects when it receives a notification, however, on the iPhone, it still shows as connected. Then, when I try to disconnect on iPhone, the whole board crashes and reboots.

biospank, 2023-06-27:
@Salman-Mirza-0 unfortunately I haven't been luck so far. My very next attempt was to fix that issue trying on a different board (esp32s3) but as you experienced it didn't work. Apart from compilation error (that has been fixed) I think the real problem comes from arduino_esp32 2.0.x library ble stack and we should investigate on that direction imho.

Salman-Mirza-0, 2023-06-28:
That is unfortunate @biospank . I am very much a beginner to this, so I am not likely to find a fix. I will do my best to explore the issue. Please let me know if you do happen to make any progress. Thanks

normen, 2023-10-16:
I also have issues getting this to work in PlatformIO with any `espressiv32` platform higher than `3.4.0` - and even then it doesn't seem to properly connect. The board is a NodeMCU 32s v2.

nguyenductoan1995, 2025-04-27:
> Xin chào, tôi đã sửa được lỗi biên dịch cho thư viện arduino_esp32 2.0.x và đã tải thành công lên bo mạch esp32-c3. Bạn có thể tìm thấy phiên bản đang hoạt động trên fork của tôi [tại đây](https://github.com/biospank/ESP32-ANCS-Notifications) Bo mạch esp32-c3 ghép nối với điện thoại của tôi nhưng không hiển thị hộp thoại cho phép cấp quyền thông báo. Tôi cũng nhận thấy trên màn hình nối tiếp rằng khi ghép nối, tôi nhận được nhật ký lỗi này:
> 
> `BT_APPL: earlier enc was not done for same device`
> 
> Có ai biết mã nào có thể viết/sửa để giải quyết vấn đề này không?

> Hi, I managed to fix compilation error for arduino_esp32 2.0.x library and successfully uploaded on a esp32-c3 board. You can find a working version on my fork [...]

I have the same problem have you solved it can you show me how to fix it pls