Smartphone-Companions/ESP32-ANCS-Notifications #28: FreeRTOS.h: No such file or directory

petoor, 2022-08-31:
Running the example i get the following error : 
`FreeRTOS.h: No such file or directory`

I can see it is called in `ancs_ble_client.h`

I am using the arduino ide

Any idea how to fix this?
`

Erchil66, 2022-09-07:
Same also looking for solution on this

petoor, 2022-09-07:
@Erchil66 I ended up using : https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/classic_bt/hfp_hf
Works perfect for my case.

closed 2022-09-07 by petoor

ManuEcheveste, 2024-04-24:
Hi @petoor, would it be a bother if you could please explain to me how you implemented your fix? As I wasn't able to figure it out.

petoor, 2024-04-24:
@ManuEcheveste have you gotten the example to run?

ManuEcheveste, 2024-04-24:
@petoor Unfortunately, no. I get the same error as you on the beginning.
`In file included from C:\Users\ManuEcheveste\Documents\Arduino\libraries\ESP32_BLE_ANCS_Notifications\src\ancs_ble_client.cpp:4:
C:\Users\ManuEcheveste\Documents\Arduino\libraries\ESP32_BLE_ANCS_Notifications\src\ancs_ble_client.h:6:10: fatal error: FreeRTOS.h: No such file or directory
 #include "FreeRTOS.h" // For asynchronous tasks
          ^~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board ESP32 Dev Module.`

Tried to look in the link that you shared, but got no idea in what to do for some time now.

petoor, 2024-04-24:
Yes I never got this repository to work. Please know that the example I'm linking to is a different repo.

ManuEcheveste, 2024-04-24:
Got it. Thank you so much for your time!
Hope this library will be updated, as it would be a dream to work with.

foxt, 2024-05-22:
I resolved the issue by downloading the latest version of the library from GitHub (uninstall the version from the library manager, download the git repo to your Documents/Arduino/Libraries folder).

However, even then it my iPhone wouldn't see it as a eligible to recieve notifications.

I used https://github.com/gabrielchristino/ANCS  instead.