Smartphone-Companions/ESP32-ANCS-Notifications #31: Connection drops after a short while and no notifications

Beutelleufel, 2023-09-29:
I use the example program "ble_connection".
The device is recognized by my cell phone without any problems. I can also connect to device.
The message `StateConnected - connected to a phone or tablet` appears.

After about 30 seconds, the message `StateDisconnected - disconnected from a phone or tablet` appears.
When I go back into the settings on my phone, I get the following error message.

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

```
Core  0 register dump:
PC      : 0x400d2ee7  PS      : 0x00060a30  A0      : 0x800d7808  A1      : 0x3ffcffc0  
A2      : 0x3ffbb29c  A3      : 0x3ffdc324  A4      : 0x80095b64  A5      : 0x3ffcd550  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x00000000  A9      : 0x80000000  
A10     : 0x007bf2f8  A11     : 0x003fffff  A12     : 0x80083c03  A13     : 0x3ffcd520  
A14     : 0x00000001  A15     : 0x3ffcdfa0  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x400900a4  LEND    : 0x400900ae  LCOUNT  : 0x00000000  

Backtrace: 0x400d2ee4:0x3ffcffc0 0x400d7805:0x3ffcffe0 0x400d503d:0x3ffd0040 0x400ea25a:0x3ffd0060 0x400eabaa:0x3ffd0080 0x40111fb1:0x3ffd00d0 0x4011411f:0x3ffd00f0
```

Thank you in advance.

fahreddinaykut, 2023-10-09:
Same problem for me. I tried with esp32-s3 and ios 15.7.9

Confused1ME1Student, 2024-01-23:
Did anyone solve this?

Byterunn3r, 2024-01-26:
Hello everyone! I am currently experiencing the exact same problem. I also use a ESP32S3 and can connect myself to it for 1-2 minutes before i get a disconnect message. im using the provided example without the hardwarepins. A run with debug-messages also doesnt reveal anything important, as far as i know.  On top of that, i cant get the ANCS Messages to work, it seems like the callback doesnt trigger or something...?
I would be really glad if this could be resolved somehow. If needed, i can provide serial monitor output and/or code used.
Thank you in advance!

Byterunn3r, 2024-02-03:
I got some more information. I have a ESP32 Devkit and with the Boardmanager version 1.0.6 it woked. Sadly, the ESP32S3 is not supported in 1.0.6 and gets added in 2.0.3.
The maker of this library is not responding to any post for close to a year so I guess this project is dead?
I cant find any other method of ANCS working so i really hope somebody finds a fix for it. I also wrote about it on Stackoverflow and Espressif Forums. I will let yall know if something changes.