Smartphone-Companions/ESP32-ANCS-Notifications #33: Trouble with ESP-32 WROOM.

Holpak, 2023-10-19:
After loading the sketch on the ESP 32 WROOM, the board goes into a loop and reloads after each attempt to start listening.

## Log:
```
20:37:13.963 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:37:13.963 -> configsip: 0, SPIWP:0xee
20:37:13.963 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:37:13.963 -> mode:DIO, clock div:1
20:37:13.963 -> load:0x3fff0030,len:1344
20:37:13.963 -> load:0x40078000,len:13964
20:37:13.963 -> load:0x40080400,len:3600
20:37:13.963 -> entry 0x400805f0
20:37:14.362 -> [    22][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
20:37:14.362 -> ESP32-ANCS-Notifications Example
20:37:14.362 -> ------------------------------------------
20:37:14.362 -> [    33][I][esp32notifications.cpp:103] begin(): [BLENotifications] begin()
20:37:15.061 -> [   725][V][BLEDevice.cpp:76] createServer(): >> createServer
20:37:15.061 -> [   725][V][BLEServer.cpp:291] registerApp(): >> registerApp - 0
20:37:15.061 -> [   726][V][FreeRTOS.cpp:189] take(): Semaphore taking: name: RegisterAppEvt (0x3ffdbdb4), owner: <N/A> for registerApp
20:37:15.061 -> [   736][V][FreeRTOS.cpp:198] take(): Semaphore taken:  name: RegisterAppEvt (0x3ffdbdb4), owner: registerApp
20:37:15.094 -> [   745][V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: RegisterAppEvt (0x3ffdbdb4), owner: registerApp for registerApp
20:37:15.094 -> [   746][D][BLEDevice.cpp:102] gattServerEventHandler(): gattServerEventHandler [esp_gatt_if: 4] ... ESP_GATTS_REG_EVT
20:37:15.094 -> [   768][V][BLEUtils.cpp:1530] dumpGattServerEvent(): GATT ServerEvent: ESP_GATTS_REG_EVT
20:37:15.127 -> [   775][V][BLEUtils.cpp:1673] dumpGattServerEvent(): [status: ESP_GATT_OK, app_id: 0]
20:37:15.127 -> [   783][V][BLEServer.cpp:144] handleGATTServerEvent(): >> handleGATTServerEvent: ESP_GATTS_REG_EVT
20:37:15.127 -> [   792][V][FreeRTOS.cpp:143] give(): Semaphore giving: name: RegisterAppEvt (0x3ffdbdb4), owner: registerApp
20:37:15.127 -> [   802][V][BLEServer.cpp:281] handleGATTServerEvent(): << handleGATTServerEvent
20:37:15.160 -> [   802][V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: RegisterAppEvt (0x3ffdbdb4), owner: <N/A>
20:37:15.160 -> [   819][V][BLEServer.cpp:295] registerApp(): << registerApp
20:37:15.160 -> [   824][V][BLEDevice.cpp:83] createServer(): << createServer
20:37:15.160 -> [   829][I][esp32notifications.cpp:144] startAdvertising(): [BLENotifications] startAdvertising()
20:37:15.194 -> [   838][I][BLEDevice.cpp:577] getAdvertising(): create advertising
20:37:15.194 -> [   844][D][BLEDevice.cpp:579] getAdvertising(): get advertising
20:37:15.194 -> [   850][V][BLEAdvertising.cpp:153] setAdvertisementData(): >> setAdvertisementData
20:37:15.194 -> [   857][V][BLEAdvertising.cpp:161] setAdvertisementData(): << setAdvertisementData
20:37:15.194 -> [   860][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_DATA_RAW_SET_COMPLETE_EVT
20:37:15.227 -> [   865][V][BLEAdvertising.cpp:187] start(): >> start: customAdvData: 1, customScanResponseData: 0
20:37:15.227 -> [   874][V][BLEUtils.cpp:1065] dumpGapEvent(): [status: 0]
20:37:15.227 -> [   883][D][BLEAdvertising.cpp:206] start(): - no services advertised
20:37:15.227 -> [   888][D][BLEDevice.cpp:579] getAdvertising(): get advertising
20:37:15.227 -> [   894][V][BLEAdvertising.cpp:252] start(): << start
20:37:15.260 -> [   900][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 4]
20:37:15.260 -> [   912][D][BLEAdvertising.cpp:529] handleGAPEvent(): handleGAPEvent [event no: 4] process default case
20:37:15.260 -> [   924][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
20:37:15.260 -> [   931][V][BLEUtils.cpp:1206] dumpGapEvent(): [status: 0]
20:37:15.293 -> [   936][D][BLEDevice.cpp:579] getAdvertising(): get advertising
20:37:15.293 -> [   942][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 1]
20:37:15.293 -> [   949][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: ESP_GAP_BLE_ADV_START_COMPLETE_EVT
20:37:15.293 -> [   958][V][BLEUtils.cpp:1074] dumpGapEvent(): [status: 0]
20:37:15.293 -> [   964][D][BLEDevice.cpp:579] getAdvertising(): get advertising
20:37:15.327 -> [   969][D][BLEAdvertising.cpp:506] handleGAPEvent(): handleGAPEvent [event no: 6]
20:37:16.137 -> ets Jul 29 2019 12:21:46
20:37:16.137 -> 

```

## Code:
```cpp
// @author James Hudson bugs.feedback.whatever@gmail.com
// @todo license text and header - please see github project for license

// A simple example program to retrieve notifications from your device, and output them to the Serial console.

// Header for this library, from https://www.github.com/Smartphone-Companions/ESP32-ANCS-Notifications.git
#include "esp32notifications.h"

/////
// Different hardware presets; uncomment the correct device.
// Or feel free to add your own board layout for your specific hardware.

//////////
// Example code begins

// Create an interface to the BLE notification library
BLENotifications notifications;

// Holds the incoming call's ID number, or zero if no notification
uint32_t incomingCallNotificationUUID;

// This callback will be called when a Bluetooth LE connection is made or broken.
// You can update the ESP 32's UI or take other action here.
void onBLEStateChanged(BLENotifications::State state) {
  switch(state) {
      case BLENotifications::StateConnected:
          Serial.println("StateConnected - connected to a phone or tablet"); 
          break;

      case BLENotifications::StateDisconnected:
          Serial.println("StateDisconnected - disconnected from a phone or tablet"); 
          /* We need to startAdvertising on disconnection, otherwise the ESP 32 will now be invisible.
          IMO it would make sense to put this in the library to happen automatically, but some people in the Espressif forums
          were requesting that they would like manual control over re-advertising.*/
          notifications.startAdvertising(); 
          break; 
  }
}

// A notification arrived from the mobile device, ie a social media notification or incoming call.
// parameters:
//  - notification: an Arduino-friendly structure containing notification information. Do not keep a
//                  pointer to this data - it will be destroyed after this function.
//  - rawNotificationData: a pointer to the underlying data. It contains the same information, but is
//                         not beginner-friendly. For advanced use-cases.
void onNotificationArrived(const ArduinoNotification * notification, const Notification * rawNotificationData) {
    Serial.print("Got notification: ");   
    Serial.println(notification->title); // The title, ie name of who sent the message
    Serial.println(notification->message); // The detail, ie "be home for dinner at 7".
    Serial.println(notification->type);  // Which app sent it
    Serial.println(notifications.getNotificationCategoryDescription(notification->category));  // ie "social media"
    Serial.println(notification->categoryCount); // How may other notifications are there from this app (ie badge number)

}

// A notification was cleared
void onNotificationRemoved(const ArduinoNotification * notification, const Notification * rawNotificationData) {
     Serial.print("Removed notification: ");   
     Serial.println(notification->title);
     Serial.println(notification->message);
     Serial.println(notification->type);  
}

// Standard Arduino function which is called once when the device first starts up
void setup() {
  // Button configuration. It is usual to have buttons configured as INPUT_PULLUP in the hardware design,
  // but check the details for your specific device 

    Serial.begin(115200);
    while(!Serial) {
        delay(10);
    }

    Serial.println("ESP32-ANCS-Notifications Example");
    Serial.println("------------------------------------------");    

    // Set up the BLENotification library
    notifications.begin("BLEConnection");
    notifications.setConnectionStateChangedCallback(onBLEStateChanged);
    notifications.setNotificationCallback(onNotificationArrived);
    notifications.setRemovedCallback(onNotificationRemoved);
}

// Standard Arduino function that is called in an endless loop after setup
void loop() {   
}

Eonassis, 2023-10-19:
no update in bib, is bad, dont run

Mincka, 2024-01-28:
Try this version, it works great for me on this board:
https://github.com/gabrielchristino/ANCS